Nate711/StanfordDoggoProject #2: Minor design flaw in CAD

zebrassimo, 2019-04-24:
Hi gents, while trying to get myself confident with the design (maybe I want to rebuild doggo :-), I found this:

[image]
blue: SDP intersects with Outer Leg Shaft
red: SDPs intersect with each other

I trust you have fixed this after printing.

CrazyAZ, 2019-04-24:
It's great to see interest in rebuilding Doggo. 

It appears that your CAD is using two inner shaft pulleys, when instead there are separate pulleys for the inner and outer shaft.  The CAD should look like this:
[image]
And the pulley that goes on the outer shaft looks like this:
[image]
As far as I can tell this is what is in our distributed full CAD, but if you still think there's an issue I can look into it.

Hope this was helpful. Let us know if you have any more questions.

zebrassimo, 2019-04-24:
Thanks Aaron! I had just downloaded the cad files from the source link (need to check, it‘s already late in Germany). The issue seems to be a version conflict between your internal files and the published ones.
    	
    	

    	Outlook für iOS beziehen

https://github.com/Nate711/StanfordDoggoProject/issues/2#issuecomment-486385538

zebrassimo, 2019-04-25:
So I figured that the link https://a360.co/2OBxTbH leads me to the browser view and here it's fine:
[image]
When I download the Fusion360 archive (email required) I receive a file called Full Doggo CAD.f3z which which is binary the same as the one I uploaded into my Fusion360 which caused the issue initially mentioned.

Well, never mind I'll get alog with this.

maxsu, 2019-08-10:
I've checked all four sets of inner and outer pulleys in the v1.0 doggo hardware model (https://a360.co/2OBxTbH), and they look correct:
[image]
[image]

This looks like it is internal revision 3:

[image]
Therefore I can only guess that the issue was caused by one of:

A) There is some earlier, publicly available revison that shows @zebrassimo's issue
B) There was some import glitch or other error on @zebrassimo's side

I believe that Fusion 360 Public Links point to the latest revision of a design, which would technically allow A) to happen. However @zebrassimo's 'same binary' comment makes me give more weight to B).

It would be useful to gain insight on the CAD environment that caused the issue. However, if there is no replication then this issue seems closed.

 @zebrassimo, what CAD software did you import into? Have you been able to replicate the issue since?