Nate711/StanfordDoggoProject #5: Voltage information

netphreak0101, 2019-05-22:
Looking though the docs and the BOM, I'm trying to see the voltage needed to drive the motor controllers. You referenced 48v on the git, but the motor drivers are only 24v. Are you driving each controller through a separate battery? Do you have a better wiring schematic?

CrazyAZ, 2019-05-23:
The ODrive motor controllers that we use support up to 48V, but can also be run on a power supply as low as 12V.  In our design we run two 24V batteries in parallel, which supply power to the entire system.

closed 2019-05-25 by Nate711

CS4Liu, 2019-07-19:
So do the 24v version of ODrive motor controllers work well too?

UPUPGOO, 2020-01-02:
> So do the 24v version of ODrive motor controllers work well too?

In my test(24 version), when you conduct the 'backflip' command, the torque is not big enough. And the doggo failed to do a backflip. So I decide to use 48v version of ODrive.