Nate711/StanfordDoggoProject #7: How did you machine the pulley brace?

mengqlTHU, 2019-05-28:
[image]
First thank you very much for making the doggo open source.
I am a little confused because this pulley brace seems to have two version: the aluminum one and the ABS plastic one. Which one are you using in the current version? And how did you assemble this part with the bearings? Thanks.

ZhangLiangShan, 2019-05-28:
According to my understanding, this is just a three-shaft positioning assembly, ensure the parallelism between the three-shaft. I used 6061 aluminum alloy processing, which is not cheap, it cost me 1100 RMB

Nate711, 2019-05-28:
Hi there, we actually 3d printed that item for the second Doggo we built. It was just cheaper and the aluminum wasn't necessary. The bearings were press-fit in with some super glue.

closed 2019-05-28 by Nate711