Nate711/StanfordDoggoProject #9: Regarding current flow amperage of each motor

HelloPlanet, 2019-06-02:
Hi Nate and team, 

I'm trying to build a doggo here, while checking ODrive part, I found that the battery have a peak working current of 75A, while banana connectors to motor is 3.5mm which allow 30A continuous current. This is a bit unclear to follow.

Do we have any test results on how much cuurent each motor draw from the drive? Thanks.

Nate711, 2019-06-03:
Hi there. Thanks for your diligence! The peak current drawn by the motors is around 40A and the average is around 5A so the 30A continuous limit on the bullet connectors is completely fine. 

For the electronics wiring, please reference the README for this project which includes a little diagram on how everything goes. Basically the two batteries are connected in parallel to the PDB, which then powers the four ODrives. It's very much like a quadcopter setup except that we have two batteries in parallel going to the PDB instead of just one.

closed 2019-06-03 by Nate711

HelloPlanet, 2019-06-03:
Thanks Nate, that explained everything.

790947648, 2019-06-04:
Looking forward to seeing your second edition of Doggo