Nate711/StanfordDoggoProject #10: About the belt model

rabbit-zc, 2019-06-03:
First thank you for making the doggo open source. 
Then,My friends and I are planning to make one.
Now I want to check if the belt is this model:

[image]
[image]

thank you
Looking forward to your reply

Nate711, 2019-06-03:
That's the one! Let me know when you get to fabricating the coaxial tubes because I might have some ideas on how to make it easier on you guys.

Nate711, 2019-06-03:
By the way if the GT2 belts are out of stock, the GT3 belts will work as well.

closed 2019-06-03 by Nate711

rabbit-zc, 2019-06-05:
OK,thank you.