Nate711/StanfordDoggoProject #11: T-Motor MN5212 Not Available

danbo2, 2019-06-04:
Hi,

The suggested T-Motor MN 5212 is currently unavailable in Amazon. Can you give another one to replace it?

Thanks,
Yiyi

HelloPlanet, 2019-06-04:
I‘m not from Project Doggo team, but as far as i know, you can find other 5212 304KV motors in stead.

The key specs you may want to check are as follows,
340KV -- this has to be specific
35A 840W max continuous power -- this could be a bit different
dims: Shaft 6mm, Stator 52mm x 12mm -- this better be specific, especially the shaft and axis size
motor 59 x 33.5mm -- this is the outer dimension and will not affect much, as long as the stator and axis is at same place, but if you had a motor much bigger in outer dimension, it may not fit in the relatively tight space inside Doggo.

No. of Cells (Lipo): 4-8S -- this will decide the working voltage range of the motor. as far as i know, all 5212 motors should be in this range.

Nate711, 2019-06-13:
Thanks HelloPlanet for answering the question! You can definitely make Doggo work with a variety of motors, but if the dimensions are different then you'll probably have to modify the mechanical design. Your best bet is to buy the T-Motor 5212 motors which you can also find at:
http://store-en.tmotor.com/goods.php?id=377
https://www.foxtechfpv.com/tmotor-mn5212-kv340.html

I hope they become available again on Amazon since they tend to be listed at a cheaper price there.

closed 2019-06-13 by Nate711

danbo2, 2019-06-13:
Thanks. It works.