Nate711/StanfordDoggoProject #14: When we clone the project, the folders are empty, could you please solve the problem?

mohakhalili71, 2019-06-12:
I want to use your project. But unfortunately, when I clone the project, I will encounter empty folders. All folders are empty, except the photos folder. Please check this problem.

Thank you for helping me.

maxsu, 2019-06-12:
This project uses git submodules, which can be initialized by performing

> git submodule update --init --recursive --remote

This populates the subdirectories:

```
~/src/StanfordDoggoProject$ tree -L 2
.
├── Doggo
│   ├── lib
│   ├── platformio.ini
│   ├── README.md
│   └── src
├── DoggoHardware
│   └── FullCAD
├── images
│   ├── belly.jpeg
│   ├── coax.jpeg
│   ├── diagram.jpeg
│   ├── electronics.jpeg
│   ├── foot.jpeg
│   ├── frame.jpeg
│   ├── hero.jpeg
│   ├── hero.jpg
│   ├── joint.png
│   ├── sides.jpeg
│   ├── trajectory.png
│   └── wiring.png
├── LICENSE
├── ODrive
│   ├── analysis
│   ├── Arduino
│   ├── CHANGELOG.md
│   ├── docs
│   ├── Firmware
│   ├── ODrive_Workspace.code-workspace
│   ├── README.md
│   └── tools
└── README.md
```

Hope this helps you get doggoing 😉

mohakhalili71, 2019-06-12:
Thank you Max, it does work now.

Nate711, 2019-06-13:
Thanks so much Maxsu for answering that question!

closed 2019-06-13 by Nate711