Nate711/StanfordDoggoProject #18: BOM map to CAD

kigerzh, 2019-06-24:
Dear Aaron,
Thanks so much for the great project. 
Now I am plan to rebuild the Doggo .  I checked the BOM ,but it is hard to map with the CAD,due to missing  uniqe component ID. Especially  for some small objects. For example in CAD,  what is  the object under encoder?  :  Full Doggo Assembly v170/single plate design v471:1/AS5048A v26:4/Component2:1 

Is it possible to  export  a BOM from the CAD  and then  you give some comment in that BOM ?   Or  add more information in the CAD file .

HelloPlanet, 2019-06-27:
As a matter of fact, i'm combing the whole CAD file to build a new BOM....

SunHR-SWJTU, 2019-07-12:
The object under AS5048A looks like the magnet of encoder.

maxsu, 2019-07-19:
I'd love to help revise the bom / make it more accessible. I'm working in Rhino. Ideas are welcome!

bestjinx, 2019-08-09:
Thank you very much for your work. Now we want to reproduce your project, but I find that there seems to be no complete circuit connection schematic in the files you uploaded, so I hope you can upload it if it's convenient. Thank you very much

YANGKAI-MAX, 2020-01-16:
I haven't found the general assembly drawing. Who can help me?