Nate711/StanfordDoggoProject #21: Wondering about the reasoning behind part selections

askewc, 2019-06-27:
Awesome project! I'm looking to take a crack at building Doggo, but as $3k is a lot for a humble hobbyist, I'm looking to cut costs/reuse parts I already have. It'd be great to know _why_ you chose certain parts, so I can understand what I'm trading off.

- Motors
  - ~$800 is a lot! Are there specific advantages to using the T-Motor MN5212 KV340 over motors with the same stator size from another brand? Are there other ways one could cut cost here?

- Legs
  - Why do they need to be aluminum? I'd guess it has something to do with handling high torque, like with the braces-- if so, are there many scenarios in which the legs would be so stressed (besides jumping / falling)? 
  - [MIT's Super Mini Cheetah](https://www.youtube.com/watch?v=CEAfiyreL9E) walked around on plastic legs. No backflips, though :)

- ODrive Controllers
  - I have some SaberTooth and RoboClaw motor controllers from some old projects. How does ODrive outperform its contemporaries?
  - (although honestly, are my controllers even comparable? They're so much fatter than the ODrive)?

Thanks again for sharing your work and taking questions!

HelloPlanet, 2019-06-27:
Just FYI, i ordered fiber glass 3D printed legs out of curiosity, and they have no match in strength to 6061 aluminium. 
Should you really would like to cost down the project, please consider reconstruct the foot linages.

As for ODrive. I think the team stated in readme that they need ODrive to have close loop control of the motor. That is, they leave the whole motor speed sensing, controlling and feedback thing to ODrive thus allow Teensy to focus on sensing and controlling the attitude of the whole Doggo.

Nate711, 2019-07-01:
Hi askewc! 

1) The motors are probably the best spot to look for cost savings, but it's not trivial to swap in a different motor because of varying  geometry. As for alternate motors, you can really use any motor with a similar torque rating, which in most cases means finding a motor with the same or larger stator geometry (5212). However, a different motor will nearly certainly have slightly different geometry, which means you will have to adjust the encoder mount to adapt to the new magnet<->encoder distance, and the pulleys to accommodate the different rotor height. 

2) The legs definitely do not have to be made of aluminum, but if you choose to make them out of 3D printed nylon for instance, you will certainly have to strengthen the parts and you may also have to come up with a solution to transferring torque between the coaxial shafts and the links that attach to them.

3) ODrive is as cheap as you can get right now for the functionality it offers. I think a truly inexpensive quadruped robot may only be achievable with servo motors. I hope someone makes that soon!

closed 2019-07-01 by Nate711