Nate711/StanfordDoggoProject #22: How to place IMU board

kigerzh, 2019-07-08:
Hi Nathan,
How to place the IMU board:  the X point to head or Y point to head?

HelloPlanet, 2019-07-09:
[image]

as you can see in this picture,
the "front" direction Doggo heading to is X, to his "left" is Y, and the direction to ground is Z.

or maybe i can misunderstood front and back, but i don't think it will matter much, as Doggo is symmetric.

CrazyAZ, 2019-07-10:
The photo of our electronics plate that @HelloPlanet posted is going to be the best reference as to how we oriented ours.  However, the direction the imu is pointing in the picture is actually with Y pointing forward.  With that said, you can orient it any way you want as long as the pitch is calculated using the right gyro axis in imu.cpp.

closed 2019-07-10 by CrazyAZ