Nate711/StanfordDoggoProject #23: The output value of the encoder fluctuates greatly！！

UPUPGOO, 2019-07-08:
Hi. It is a nice job. I am trying to make a doggo, but I meet a problem. I am using AS5047P as the encoder, but the output value of the encoder changes a lot. For example, if I set the position at 1000, the output value fluctuates around 1000 and the fluctuation range is around 50. I find the motor jittering. Is it normal? How can I stop it? Now, I am ready to buy the shielded wire. Maybe, it will work. I need your help.

HelloPlanet, 2019-07-09:
Please consider post a photo of your setup.
As per AS5047P manual said, the space between the magnet and sensor chip should be 0.5-3mm.

CrazyAZ, 2019-07-10:
Unfortunately, without more information this is going to be hard to diagnose. One thing I would check is if these fluctuations occur when the motor is not being controlled.  If not, it may be that your kp gain for the PID control is too high. This would cause the motor to constantly over shoot its target and jitter.

closed 2019-12-05 by Nate711