Nate711/StanfordDoggoProject #25: Custom ODrive Firmware build with binary protocol?

HelloPlanet, 2019-07-09:
Hi Nate and team,

I've cloned your fork of ODrive software from this repo, and was trying to build the "custom binary protocol firmware" and flash it to my ODrives.

Upon building, the config file says 
`CONFIG_UART_PROTOCOL=ascii`

do i need to change this, or should i just build and flash with default config?

Thanks.

CrazyAZ, 2019-07-10:
The uart protocol being set to ascii is indeed correct. Our custom communication protocol transmits the values to the odrive as if they are ascii characters, and then our firmware knows to interpret the binary representations of those characters as floats instead.

It's been a while, but I believe we did this as a way to allow the sending multiple values at a time.  The ascii protocol is far less strict about the size of the messages, giving us more flexibility than the native binary protocol.  However, because we send the values in their binary form instead of as ascii characters it is significantly faster than the original ascii protocol.

closed 2019-07-10 by CrazyAZ

HelloPlanet, 2019-07-10:
Thanks for your detailed explanation!

mengqlTHU, 2019-07-30:
@HelloPlanet Have you successfully run the code? I think I have compiled and uploaded the ODrive and Teensy firmware correctly, but when I send command to teensy, the motor just does not move at all.