Nate711/StanfordDoggoProject #29: Error when open the fusion model version2

CS4Liu, 2019-07-19:
Hello,I try to use Autodesk Fusion 360 to open the fusion model version2 for more details of Doggo.But when I upload the 'Full Doggo Assembly.f3z' in the Autodesk Fusion 360 the error comes:
 
[image]
Do know how can I solve this problem？
Thanks!

closed 2019-07-31 by CS4Liu

CrazyAZ, 2019-07-31:
I see that you have closed this issue. Did you find a solution to it? Someone else is having the same problem, but I cannot find anything wrong in the CAD.

CS4Liu, 2019-07-31:
I didn't find solution to it. There still error when uploading the version2.Instead, I just download the Inventor 2018 version of the model, then open the '.ipt' file using Autodesk fusion 360 software.
then it works.

And I think there is some wrong with the model v172  in https://a360.co/2CPZoto because it has became like this now:

maxsu, 2019-09-18:
@CS4Liu, @CrazyAZ - this is still an issue with the project. It's caused by corrupted Simstudies: See the last post of [#34](https://github.com/Nate711/StanfordDoggoProject/issues/34), as well as this fusion 360 [support thread](https://forums.autodesk.com/t5/fusion-360-support/stanford-doggo-public-link-upload-issue/m-p/9006809/highlight/true) for a repaired version of the file.

I'd like to get this issue resolved. Can I join the CAD team?