Nate711/StanfordDoggoProject #33: Wiring Schematics Redux

maxsu, 2019-08-10:
In issue #18 @bestjinx asks whether there is a circuit schematic for doggo. In issue #20, @helloplanet contributed a tentative pinout. I'd like to address the lack of schematic, and propose a way forward towards a community contributed schematic.

In particular, I'd like to entertain the theory that:
1. There has simply been no need for a formal cad schematic so far / a schematic has been considered out of scope
    - The system uses a breadboard, and has a pretty simple communication pin layout which is encoded in the firmware
    - This would be clear to the designers and people conducting a serious review
    - It would however be a great benefit for hobbyists / students
    - But! it would be a _substantial_ labor
    - As such it may be out of scope for doggo designers' current goals

Continuing this line of thought:

2. We may have an opportunity to contribute a cad diagram as a community. Some notes towards this:
  a) The teensy's pinouts for the [AS5047D-TS_EK_AB  IMU](https://www.mouser.com/ProductDetail/ams/AS5047D-TS_EK_AB?qs=sGAEpiMZZMvc81WFyF5Eds%252bEyjFAfuLY9T1fn1h2lMw%3D) are mapped [here](https://github.com/Nate711/Doggo/blob/2e710121f618b006577a700c81c319086a46a366/src/config.h#L44-L48). Note that the current generation doggo BOM does not include the IMU.
b) Teensy -> Odrive serial ports get mapped [here](https://github.com/Nate711/Doggo/blob/32adf4e5ffb48af696df22e234c95e03d725bcf6/src/globals.cpp#L13-L16).
 c) The Xbee is connected to the Teensy via the teensy's USB host port, and simply relays USB serial commands to the Teensy (see [this](https://github.com/Nate711/Doggo/blob/master/src/usb_serial.h))
 
3. Last, but certainly not least, the electrical diagram we find in the [Doggo Paper](https://arxiv.org/ftp/arxiv/papers/1905/1905.04254.pdf) shows us that b) and c) are the only data connections in doggo. The rest of the connections are power connections that we can infer with relative ease.

[image]

A formal CAD model would be a crucial step in reducing hobbyist/student expertise barriers, and would open the way for a fully integrated open source circuit board with no breadboard component.

@bestjinx, @kigerzh, @HelloPlanet  - Would you three be interested in helping an effort to reverse engineer the circuit schematic, if necessary?

@Nate711 @CrazyAZ  - what would you guys recommend here? Can we get your blessings on an open schematic for doggo?

maxsu, 2019-08-10:
Next steps:

1. Agree on a circuit design software (eg, KiCad, Autodesk Eagle, etc.)
2. Add part IDs to the electronic parts in the BOM
3. Import component pinouts into PCB-CAD
4. Build the component netlist
5. Layout and annotate the schematic, title block, etc
6. Version and commit the schematic to the project.

Nate711, 2019-08-27:
Hi Maxsu, thanks for opening this issue. If you guys want to make a schematic we'd love it and would happily check it. We are unfortunately maybe too busy to create as nice a circuit design as you might like. However we could make a pin-to-pin table of connections if that would suffice.

niranjan1997, 2020-06-05:
Hi Nate, it would be great if you can provide the pin  to pin table