Nate711/StanfordDoggoProject #34: Issue with V2.0 CAD

maxsu, 2019-08-14:
Hi all,

It looks like Doggo's v2 CAD model is experiencing some issues - From the web viewer:

[image]

[image]

Description:
- There are left-translated copies of the right motor & shaft assemblies and the forward sheet metal component
- Affected items are translated forward relative to the side plates and legs
- Affected items are canted 3 degrees WRT the right side plate, so that their right mating faces are vertical
- All other parts and sub-assemblies appear properly positioned 

[image]
[image]

The DXF export of the v2 cad shows the same issues.

I could rebuild the model's assembly tree if necessary. Before I do that:
- Can anyone else confirm the issue?
- If the issue is more than an A360 cache glitch, is a repaired model available?

mseedman, 2019-08-14:
I can confirm the issue.  Michael

Nate711, 2019-08-27:
Can you try checking again? I re-computed the model. I really have no idea why Fusion does this....

maxsu, 2019-09-02:
I've made a [post](https://forums.autodesk.com/t5/fusion-360-support/stanford-doggo-public-link-upload-issue/td-p/9000575) on the fusion 360 forums - the community may shed some light on the upload issue.

I've also found a temporary workaround. cc: @mseedman - you may find this useful.

### Workaround: Import an Inventor 2018 Model

1. Open doggo v2's [public link](https://a360.co/2CPZoto)
2. Request, Download and Unzip the Inventor 2018 model
3. Open `Full Doggo Assembly.iam` in Fusion 360
4. Fusion will prompt you to bulk-select the rest of Doggo's files

This is useful for locally inspecting the model. One issue: this doesn't add individual parts or assemblies to your project - it bundles all of doggo into a single top level assembly.

maxsu, 2019-09-07:
Good news! Autodesk support team has provided a provisional fix for the "access denied issue" (see [this thread](https://forums.autodesk.com/t5/fusion-360-support/stanford-doggo-public-link-upload-issue/m-p/9006809/highlight/true#M65178)).  The `f3z` file contained in Jeff.Strather's `Full Doggo Assembly_Repaired.zip` loads without causing the "access denied" error.

Upon inspection, it looks like they removed the two simstudies from the project's `.f3z` file. Jeff has requested a member of the Doggo team to discuss steps to fix the original model (see the thread).

(Security note: Jeff appears to be an official member of the Autodesk team, and the file [checks out](https://www.virustotal.com/gui/file/7b64bfddaeec573c237a05f7cec86e4d40eb3dcb6a20d5a1e902fcf30e65f8c2/detection) on VirusTotal. Use your discretion here, and wait for the definitive fix if there are any concerns.)

CC: @CrazyAZ ,  @mseedman