Nate711/StanfordDoggoProject #35: about BOM spreadsheet

youye9, 2019-08-21:
Hello, thank you very much for providing all the materials of your wonderful work.
My friend and i are going to follow you to make a Doggo, but now we meet some problems.
1.We found that the V3.5 ODrives is not available in online shop, so we plan to buy V3.6 ODrives. Which voltage version is better? 56V or 24V? Should i buy it with connectors?
2.When i checked your BOM spreadsheet, i found that there were two 2mW Wire Antenna. But in README.md you said you used a 5mW Xbee. Which one is correct?
3.The product model of minitactor in BOM spreadsheet is P115Bda while in README.md is P105, which one is correct?
4.I can't find the BNO080 IMU in BOM spreadsheet, but i believed it was used in Doggo and you forgot to write it in the BOM spreadsheet, right?
Thank you very much!

Nate711, 2019-08-27:
Hi youye,
1. 24V is ok. It's better to buy without connectors. 
2. Either Xbee is ok. Just make sure you buy two of the same kind. The 2mW is preferred.
3. 115Bda is correct!
4. Thanks, I added it to the BOM. It's optional though, we only used it for backflips.

closed 2019-08-27 by Nate711

youye9, 2019-08-30:
> Hi youye,
> 
> 1. 24V is ok. It's better to buy without connectors.
> 2. Either Xbee is ok. Just make sure you buy two of the same kind. The 2mW is preferred.
> 3. 115Bda is correct!
> 4. Thanks, I added it to the BOM. It's optional though, we only used it for backflips.

Thank you very much~