Nate711/StanfordDoggoProject #37: Issue with StanfordDoggoOuterBigPulley.stl.step

ulaladungdung, 2019-09-11:
Hi!! Thanks for sharing the doggo project to the public. My team is currently facing difficulty in printing `StanfordDoggoOuterBigPulley.stl.step` as the file appears to be corrupted. Do you have any advice regarding this?
[image]
I used the download link from the shared BOM

methylDragon, 2019-09-11:
I'd like to add that I am facing a similar issue with the part.

But it also seems that the CAD file for that particular item shows a shaft instead of a 48 tooth gear?

[image]

maxsu, 2019-09-16:
@ulaladungdung, @methylDragon, see my feature branch for fresh STL and STEP files:

https://github.com/maxsu/StanfordDoggoProject/tree/feature/track-CAD-Files/DoggoHardware/3D%20Printed

These include all 3d printed parts in project.

maxsu, 2019-09-16:
Notes:

1. Most files are named in accordance to their part names in the CAD model, which (currently) disagree somewhat with the BOM. Hopefully this does not cause too much confusion.

2. Caution: BOM specifies [94180A333](https://www.mcmaster.com/94180A333) heat-set inserts for `StanfordDoggoOuterBigPulley`, however the CAD model is configured to use [94510A310](https://www.mcmaster.com/94510A310) screw-to-expand inserts. The CAD model requires slight revision before it can receive A333 inserts, so I recommend A310 inserts for now. Let me know if you want me to fast track a heat-set revision to you.

ulaladungdung, 2019-09-16:
Hi @maxsu Thanks so much for the help!!

And yes please, the revision would be amazing as we already have the A333 on hand. Cheers!!

maxsu, 2019-09-17:
See the [updated stl](https://github.com/maxsu/StanfordDoggoProject/blob/5ca2fe981353bd918723352c2b82fe12936ca445/DoggoHardware/3D%20Printed/StanfordDoggoOuterBigPulley.stl). Fusion version is [here](https://a360.co/31twOWO).

Be careful; I increased the minimum wall thickness by 15% (to 1.15mm), but the insert's centerline could drift during heated installation. Let me know if you need a heavier revision.

@Nate711:
Is there any other A333 version of the pulley?
Is a 1.15mm wall thickness suitable?

[image]

maxsu, 2019-09-17:
I reduced the height of the pulley's shoulder boss by 0.5 mm in order to accommodate the larger diameter of the A333 without altering any other parts. However,  the spacing between pulley and housing could be a problem:

[image]

I did a little exploration, and looks like we can restore the spacing if we aggressively recess the insert into the pulley face:
[image]

This also increases the minimum wall thickness:
[image]

Take care however; this requires a 1.4 mm translation of the outer shaft locating holes.

Nate711, 2019-10-20:
Our robot uses the A310 screw-to-expand inserts but that caused an issue where if you really tighten the set screws that go inside of the inserts and then try to unscrew those set screws, the A310 insert sometimes spins inside the plastic instead of allowing the set screw to unscrew.