Nate711/StanfordDoggoProject #38: Revisions: V2 Motor & Shaft Assemblies

maxsu, 2019-09-15:
There are a few minor issues in the Shaft, Leg, and Motor assemblies. They range from minor disagreement with the BOM, to a construction tree error.

I'll start with the simplest first:

## Leg Assembly:
- [ ] Set Screws missing from assembly.
       - Part #: `94180A333` Cup Point Set Screw
       - This item is present in BOM, but not assembly.

## Motor Assembly:

- [ ] `Pulley Bracket` missing from BOM
       - This part is present in assembly, but not BOM.
       - Part material needs specification: 3dp or Aluminum

- [ ] Lower AS5047 needs mounting nuts 
       [image]

- [ ] Misaligned bolt in upper AS5047 plate
       Part:  `single plate design > Screws  > 92095A181 (1):17`
   [image]

- [ ] Phantom component: `Pulley Brace (Aluminum):1 & 2`
       [image]

- [ ] Phantom bodies
       Parts: `REV 2 Small Pulley Print`, `REV 2 Small Pulley LONG Print`
       Bodies: 1 & 2

## Coaxial Assembly
[image]

- [ ] 1. Inner shaft: Unintended hole
- [ ] 2. Outer shaft: Unintended 0.5mm web
- [ ] 3. Inner shaft: Missing dowel pin
           - This part is present in BOM, but not assembly
           - Part number: `93600A455`
           - Needs interference fit sizing

  - Suggested Fix (file available upon request):
     - 1. I've removed the inner shaft from the extrude feature which creates the outer shaft's set screw hole
     - 2. I've perform the outer shaft set screw feature after the outer shaft revolve feature which creates the bearing shoulder
     - 3. I've added a dowel pin hole hole in the inner shaft, and extend the inner shaft shaft 2 mm past the pulley body as a part holding feature during dowel pin installation
[image]

[ ] `Outer Big Pulley`: BOM collision
    - Assembly has: `94510A310` (Brass Screw-to-Expand Inserts)
    - BOM has: `94180A333` (Heat-Set Inserts)

[ ] Inner shaft spacer missing from BOM
    - Assembly Name: `Spacer (1)`
    - Needs material spec: 3dp or 16.6x12.6x1 Steel washer

[ ] Duplicate Component: `SDP_a_6a53m048nf0908` and  `Rev 2 SDP_a_6a53m048nf0908:1` are identical

Nate711, 2019-09-21:
I think I fixed the CAD issues. Check it out when you have a chance!

maxsu, 2019-09-24:
@Nate711 I thank you for your response and continued work on Doggo. I'm sad to say that the cad model still experiences the `access denied` error (featured in #31, #29, #34):
[image]

This error means nobody can import doggo's design from its public link. It's really important that we get that problem fixed and restore public access to doggo's design. Thanks!

Fusion 360 support identify the problem in this F360 [forum link](https://forums.autodesk.com/t5/fusion-360-support/stanford-doggo-public-link-upload-issue/m-p/9006809/highlight/true) :
- `Single Plate Design` and `Leg Linkage Assembly` become corrupt in translation to `.f3z`
- In the short term, removing the simstudies appears to fix the model
- Support has asked for feedback to fix the real problem

Nate711, 2019-10-04:
@maxsu Hi Max, thanks for specifying which parts were corrupt. I had seem the issue before but didn't know which simulations to delete. I got the downloaded fusion archive to upload properly on my computer. Can you check too? Thanks

maxsu, 2019-10-06:
Unfortunately I'm still having the same issue. I used the latest url from `DoggoHardware/FullCAD`, which is https://a360.co/30j7ANn

The archive has the two bad simstudies present in previous versions of the file:
`488f29e2-8a9f-450c-9303-3dc281e19bad.simstudy`
`2a44713a-4246-48b2-9885-3395ac1c1014.simstudy`

Perhaps you are looking at a different URL? Can you check if your archive has simstudies?

 1. Rename the `full CF.f3z` into `full CF.zip`
 2. Open the zip and check for the simstudies above

Nate711, 2019-10-08:
@maxsu So I see that the simstudies are still present. But I've deleted them in both Fusion 360 parts. Is there anything left I can do?

maxsu, 2019-10-10:
@Nate711 sorry for my late response. You can talk to F360 support member jeff.strater [here](https://forums.autodesk.com/t5/fusion-360-support/stanford-doggo-public-link-upload-issue/m-p/9006809/highlight/true). I'd let him know you've tried deleting the sim-studies with no luck - he can work with the F360 dev team to get to the bottom of the issue.