Nate711/StanfordDoggoProject #43: Motor Spacer

danbo2, 2019-10-25:
Can someone tell me whether the Motor Spacer is in purchasing list.
[image]

Kevinskwk, 2019-11-13:
Hi there! I also didn't find this spacer in the purchasing list. Alternatively, we laser cut the spacer out of acrylic sheet when we made it. I don't actually recommend using acrylic since this part seems to be load-bearing and this is only our temporary solution. I think you should consider using stronger material if possible. Cheers!

danbo2, 2019-11-19:
@Kevinskwk  Thanks for your help. I have downloaded the fusion 3D model and extracted this part for 3D printing.

closed 2019-11-19 by danbo2