Nate711/StanfordDoggoProject #44: How to let the Motor Run？ I just want test the system can run success？

xuankaicheng, 2019-10-28:
Dear Nate711
I just want test the system can run success，next is my operating。But the motor can't run.
Witch of the parameters is necessary to let the motor run?
I use one odrive3.5-24V(Flash firmware success)(tup.config)
[image]

One motor（is MN5212-KV340） connent to M0,one as5047d lock on moto and connect the M0 ABZ. use the  power source is 150W。I send 'H' command(Teensy3.5 can receive the command)。
Odrive connet Teensy3.5 with UART1。
But the  motor don't run. Some of parameters which should i set can let the motor run.

AS5047D  Lock on the motor The ABI connect the ODrive M0 ABZ;Motro connect M0。
[image]

AS5047D I PIN is always low
the AB pin wave in picture。
[image]

 I use the Teensy3.5(#define IMU_VERBOSE 1) power on,Receive the message
[image]

next send HOP command，receive the message
[image]

Thank You!!!

xuankaicheng, 2019-10-30:
I find the answer use doggo_setup.py script,I will try it.

xuankaicheng, 2019-10-31:
Hi Nate711
I use doggo_setup.py configure the odrive successs。Power On calibration motors and encoders success the motors can run.
(Teensy3.5) RX1 -- (Odrive3.5-24V )GPIO1
(Teensy3.5) TX1 -- (Odrive3.5-24V )GPIO2
Teensy3.5 connect Odrive success(I USE multimeter test it)
I use serials tool send command H,receive Teensy3.5 message,but motors can,t run.
(Question Teensy3.5 how to configure can let motors run witch command).
The power outout 24V  5A。

[image]

Another question is the BOM display the odrive is 24V version,but the github say you use the ODrive is 48 whitch is the right version of ODrive.

closed 2019-11-11 by xuankaicheng

caobencheng, 2019-11-14:
Hi Nate711, and xuankaicheng, I met the same problem just as xuankaicheng met. I tested the Odrive board with doggo_setup.py, and it looks good. However, I connect the Odrive and the Teensy board on UART 1, and I run the Teensy, I found the motor just not run and no any response. I use oscilloscope to check the signals on Pin0(RX1) and Pin1(TX1) of Teensy, the signals keep high without any changing. So there is no UART data sent on Pin1 and no data received on Pin0. What is the reason that no data sent on UART 1 of Teensy? thanks. ben

niranjan1997, 2020-09-26:
Hey @xuankaicheng , did you solve this issue ? even I have the same issue please help me fix this. thank you.