Nate711/StanfordDoggoProject #46: No data sent on Uart1 to Odrive

caobencheng, 2019-11-14:
Hi, Nate711, and the team, I met a problem and do not know why.I have tested the Odrive board with doggo_setup.py, and it looks good. I connected one Odrive board with one motor and the Teensy board through Uart1, then power on. I run the Teensy, and then it started with the led flashing and command line prompts normally. However, the motor has no any movings 
or actions. I send command H, or T, or others, still the motor has no response. I use oscilloscope to check the signals on Pin0(RX1) and Pin1(TX1) of Teensy, the signals keep high without any changing all the time. So I can confirm that there is no UART data sent on Pin1 and no data received on Pin0. What is the reason that no data sent on UART 1 of Teensy? Thanks. BEN

caobencheng, 2019-11-15:
Hi, Nate711, and the team, I have solved the problem. Right now the TEENSY can successfully control the odrive board and the motor can run according to the E(dancing), T(Trot), R(reset) and S(stop) commands. Thanks. BEN

closed 2019-12-05 by Nate711

UncleMask, 2020-07-23:
I have the same problem, and my LED is not bright, I was wondering how you solve this problem

caobencheng, 2020-07-23:
Hi, UncleMask, I found the led port type will be changed by IMU function failing,so I  move the calling of "BlinkThread" at last in  chSetup(). You can try.

UncleMask, 2020-07-24:
Yes, I solved the problem after seeing your another issue. Thanks very much

niranjan1997, 2020-10-03:
> Hi, Nate711, and the team, I have solved the problem. Right now the TEENSY can successfully control the odrive board and the motor can run according to the E(dancing), T(Trot), R(reset) and S(stop) command. Thanks. BEN
Hello @caobencheng, I have the same exact problem, may I know how did you resolve this issue. Thanks in advance.

qw7798965, 2021-01-08:
Hello @caobencheng, I have the same exact problem, may I know how did you resolve this issue. Thanks in advance.