Nate711/StanfordDoggoProject #50: Teensy serial monitor is blank

niranjan1997, 2020-09-25:
Hello Nate, I have tried building the firmware using Platformio on both Vs code and Atom. The build was successful without errors and uploaded to the board. Even after successful build and upload. Nothing shows up in the serial monitor, and the led on board is not blinking. what might be the cause? please help me fix this. I am on the verge of completing this project.

RajPShinde, 2020-09-26:
If you are not using the Xbee, comment #define USE_XBEE in config.h

niranjan1997, 2020-09-26:
yes, I am using Xbee but according to the code, the Led on the teensy board should glow if the code is properly uploaded to the board right? The led on my teensy isn't glowing even after successful build and upload.

closed 2020-10-03 by niranjan1997