Nate711/StanfordDoggoProject #51: Belts Skipping teeth

RajPShinde, 2020-09-30:
I just completed the assembly. The issue is the belts are skipping teeth when I try to Trot using default parameters. The robot does stand on its own without any issue, the skipping only takes place at high velocities. I am using the provided 3D Printed Brace and HTD 3M Belts. Can you please help me?

akrolic, 2021-06-19:
Hey @RajPShinde,

I hope you have been able to resolve this issue, but I can offer some advice from our teams experience. We were able to tension the belts by loosening the drive shaft assembly and pulling it away from the motors. With it pulled away and the belts in tension, you can have someone else  tighten the assembly in place. This has worked for now, a future improvement would be to ensure that the assembly will not shift from the final tensioned position. 

If your issue is not that the belt isn’t tensioned enough, then perhaps your belts could be misaligned. We solved our misalignment by adding appropriate spacing to the distance between the two belts. We also varied the height of the drive shaft a bit using washers to get a perfect alignment. 

Let me know if you need any help!

closed 2022-04-20 by RajPShinde

daydayupzhang, 2022-04-20:
你好，我已经收到你的来信。