Nate711/StanfordDoggoProject #52: Motor movement failed with

DBAjust115, 2021-01-05:
The odrive arduinotest test can run the motor successfully. After burning the whole file, the test cannot be carried out (I tested a single odrive with two motors, without IMU module). Is it possible that the IMU module needs to be added?

ducktrA, 2021-03-04:
Hi i am chasing you and just flash the teensy. I dont have the solution. But you are not alone.

ducktrA, 2021-03-12:
@DBAjust115 how did you compile the Odrive? Did you use an older toolchain?

I am running into a problem when using std::enable_if

*  4) arm-none-eabi-g++ -std=c++14 -c communication/communication.cpp -DHW_VERSION_MAJOR=3 -DHW_VERSION_MINOR=5 -DHW_VERSION_VOLTAGE=24 -DUSB_PROTOCOL_NATIVE -DUART_PROTOCOL_ASCII -D__weak="__attribute__((weak))" -D__packed="__attribute__((__packed__))" -DUSE_HAL_DRIVER -DSTM32F405xx -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Wall -Wfloat-conversion -fdata-sections -ffunction-sections -g -gdwarf-2 -Og -ffast-math -fno-finite-math-only -IDrivers/DRV8301 -IMotorControl -Ifibre/cpp/include -I. -IBoard/v3/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -IBoard/v3/Middlewares/Third_Party/FreeRTOS/Source/include -IBoard/v3/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -IBoard/v3/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -IBoard/v3/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -IBoard/v3/Drivers/STM32F4xx_HAL_Driver/Inc -IBoard/v3/Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -IBoard/v3/Drivers/CMSIS/Device/ST/STM32F4xx/Include -IBoard/v3/Drivers/CMSIS/Include -IBoard/v3/Inc -I. -IDrivers/DRV8301  -o build/communication_communication.cpp.o
In file included from /usr/include/newlib/c++/9.2.1/bits/move.h:55,
                 from /usr/include/newlib/c++/9.2.1/bits/stl_function.h:60,
                 from /usr/include/newlib/c++/9.2.1/functional:49,
                 from communication/communication.h:9,
                 from communication/communication.cpp:4:
/usr/include/newlib/c++/9.2.1/type_traits: In substitution of 'template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Cond = (0 == 1); _Tp = void]':
fibre/cpp/include/fibre/protocol.hpp:948:5:   required from 'class ProtocolFunction<Controller, std::tuple<float, float, float>, std::tuple<> >'
MotorControl/controller.hpp:158:39:   required from here
/usr/include/newlib/c++/9.2.1/type_traits:2384:11: error: no type named 'type' in 'struct std::enable_if<false, void>'
 2384 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
      |           ^~~~~~~~~~~
 *** tup messages ***
 *** Command ID=526 failed with return value 1
 [   ETA~=<1s   ]  71%
 *** tup: 10 jobs failed.
make: *** [Makefile:14: all] Error 1